Every-Flavor-Robotics/motorgo-mini-driver #32: WiFi connection string literal - no spaces or single quotes

stilsota, 2024-01-02:
connecting to wifi SSID with a space or " ' " in the name (single quote) throws two different errors. Not sure if this applies to the password as well, but likely need a way to handle special characters, spaces, and other things that can throw off reading the string. 

example output from building the project while connecting to SSID = "Tom's iPhone (2)\"

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-motorgo-mini.html
PLATFORM: Espressif 32 (6.4.0) > MotorGo Mini, Espressif ESP32-S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
ValueError: No closing quotation:
  File "/Users/tstilson/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/tstilson/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/tstilson/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/tstilson/.platformio/platforms/espressif32/builder/main.py", line 346:
    target_elf = env.BuildProgram()
  File "/Users/tstilson/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/Users/tstilson/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/Users/tstilson/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/Users/tstilson/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 118:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "/Users/tstilson/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/Users/tstilson/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 228:
    env.Append(**env.ParseFlagsExtended(flags))
  File "/Users/tstilson/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/Users/tstilson/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 190:
    for key, value in env.ParseFlags(str(raw)).items():
  File "/Users/tstilson/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Environment.py", line 1035:
    do_parse(arg)
  File "/Users/tstilson/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Environment.py", line 917:
    params = shlex.split(arg)
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/shlex.py", line 315:
    return list(lex)
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/shlex.py", line 300:
    token = self.get_token()
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/shlex.py", line 109:
    raw = self.read_token()
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/shlex.py", line 191:
    raise ValueError("No closing quotation")
```

SwapnilPande, 2024-01-03:
This looks like you're providing the SSID in the `platformio.ini` file. This is most likely just a syntax error as you need to escape the special character. Try passing "Tom\'s iPhone (2)".

If this doesn't work, please provide the code snippet that causes this to fail.

SwapnilPande, 2024-02-22:
I'm closing this issue for now, if this remains an issue please re-open

closed 2024-02-22 by SwapnilPande