Every-Flavor-Robotics/motorgo-mini-driver #111: Main branch in PIO - code fails due to configurable.h dependency

stilsota, 2024-04-11:
trying to compile "calibrate 2 motors" after pulling latest version of Main. Running in VS code. 

Following error message: 

`In file included from .pio/libdeps/calibrate_motors/MotorGo Mini Driver/src/pid_manager.cpp:1:
.pio/libdeps/calibrate_motors/MotorGo Mini Driver/include/pid_manager.h:7:10: fatal error: configurable.h: No such file or directory

**********************************************************************
* Looking for configurable.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:configurable.h"
* Web  > https://registry.platformio.org/search?q=header:configurable.h
*
**********************************************************************

 #include "configurable.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
Indexing .pio/build/calibrate_motors/lib632/libSimpleFOCDrivers.a
*** [.pio/build/calibrate_motors/lib768/MotorGo Mini Driver/pid_manager.cpp.o] Error 1
================================================ [FAILED] Took 24.75 seconds ================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `

SwapnilPande, 2024-04-11:
Duplicate of #107

SwapnilPande, 2024-05-13:
PIO Version 6.1.15 seems to have fixed the bug that causes this issue. We should verify on all platforms and make sure the library builds correctly now.

SwapnilPande, 2024-05-14:
This has been resolved, closing

closed 2024-05-14 by SwapnilPande